mattdibi/yampad #6: Process to print the case

Zeioth, 2020-07-10:
Please, could you explain the process to print the case? I uploaded  the files to [sculpteo](https://www.sculpteo.com)

    yampad_v2-Eco2.User-screws.dxf
    yampad_v2-Edge.Cuts.dxf

But I'm not sure how to proceed. I must print both files?
(I'm on linux so I don't have autocad to open the other 2 files)
Thank you.

Zeioth, 2020-07-10:
Ok after tweaking with fussion (web), I can see there are 2 versions:

V2: Made of 4 pieces.
V3: Made of 2 pieces (wooden base and steel top plate)

Could you please upload the pieces invidivially in .dxf, so I can order them, or know a linux program I can use to do it myself? Thank you.

Zeioth, 2020-07-10:
Ok got it, 

Backplate: "yampad_v2-Edge.Cuts.dxf"
Top plate: "yampad_v2-Eco2.User-screws.dxf" -> But you must edit it on freecad before to send it for printing, to eliminate the arduino placeholder.

mattdibi, 2020-07-12:
Hi,

the .dxf are the outlines of the PCB and are there to help with the process of designing a new case for the Yampad. They can't be printed as is.

If you want a case you can start from the .f3d files which can be edited in Fusion360. Keep in mind that those two projects were thought to be CNC'd and not 3d printed though.

Zeioth, 2020-07-14:
Do you have any file dxf file with holes for stabilizers? I'm designing a case in steel, but I'm having a bad time with that. Cheers.

Zeioth, 2020-07-17:
This is still WIP but I'm designing a new cover in metal. Might post a pull request in the future when I have it in my hands.
[test.zip](https://github.com/mattdibi/yampad/files/4938210/test.zip)

Pretty excited, I just received my PCBs
[image]

mattdibi, 2020-07-19:
> Do you have any dxf file with holes for stabilizers? I'm designing a case in steel, but I'm having a bad time with that. Cheers.

@Zeioth You should be able to extract that file from the Kicad project file by selecting the correct layer. I'll try and extract it for you but I have very little free time lately.

Zeioth, 2020-07-20:
Thanks! Don't worry I managed to do it, I'm already making sure everything is correct before to send it for printing.

closed 2020-12-07 by mattdibi