mattdibi/yampad #9: Firmware won't compile

Daveyr, 2020-08-19:
Hi there,
I'm just trying to flash the microcontroller prior to populating the pcb and have come across a problem. I have set up my qmk environment in Ubuntu, cloned your fork of the qmk firmware and entered the command,

```
make yampad:default
```
However, I receive the following error:
make: *** No rule to make target 'yampad:default'.  Stop.

Please can you help? Looking at the keyboards subfolder there is no yampad folder within, in the latest commit.

Daveyr, 2020-08-19:
The yampad folder isn't present on master when cloned. However, it *is* there on a branch called yampad. Is the yampad branch default when you view it on Github and master is default when cloning? 

Very confusing for someone who doesn't know git! Switching to yampad branch doesn't change the error message though.

mattdibi, 2020-08-19:
Hi,
yeah the yampad firmware lives only in my fork for now. Not yet pushed to the QMK main repository.
Regarding your error, after switching to the [yampad branch](https://github.com/mattdibi/qmk_firmware/tree/yampad) everything should work. Could you paste here the commands you issued for building?

Daveyr, 2020-08-20:
I can do one better and issue a merge request to the QMK-firmware repo if that works? I forked a recent version so it won't be too difficult. The steps I took are simplified below.

Set up my QMK environment in Linux as per https://docs.qmk.fm/#/newbs_getting_started
In a temp folder and using a terminal type, `git clone https://github.com/mattdibi/qmk_firmware`. We can delete this when finished.
`cd qmk_firmware` to change directory to qmk_firmware
Then `git checkout yampad`
Copy ./keyboards/yampad to ~/qmk_firmware/keyboards (this is the default location when setting up qmk in step 1)
Type `cd ~/qmk_firmware` to change directory there
Type `qmk compile -kb yampad -km default`
Type `make yampad:default:avrdude`

The steps above allowed me to successfully flash my pro micro. I haven't tested in a numpad yet. If I can arrange a merge with the original qmk_firmware then all other users would only have to do the last three steps.

Daveyr, 2020-10-12:
Steps above verified working on a real numpad.

nbaptist16, 2021-02-22:
I followed @Daveyr 's instructions and was able to compile, but am having trouble connecting to the programmer... this is the terminal output:

`nckmlb@xpso:~/qmk_firmware$ make yampad:default:avrdude
QMK Firmware 0.11.68
Making yampad with keymap default and target avrdude

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  25244	      0	  25244	   629c	.build/yampad_default.hex

Copying yampad_default.hex to qmk_firmware folder                                                   [OK]
Checking file size of yampad_default.hex                                                            [OK]
 * The firmware size is fine - 25244/28672 (88%, 3428 bytes free)
Detecting USB port, reset your controller now.....................................................................................................................
Device /dev/ttyACM0 has appeared; assuming it is the controller.
Waiting for /dev/ttyACM0 to become writable..

Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding

avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
^Cmake[1]: *** [tmk_core/avr.mk:193: avrdude] Interrupt
make: *** [Makefile:522: yampad:default:avrdude] Interrupt
`

mattdibi, 2021-02-22:
@nbaptist16 Could [this](https://arduino.stackexchange.com/a/75189) be the issue? (Not firmware related BTW)

nbaptist16, 2021-02-23:
> Detecting USB port, reset your controller now.....................................................................................................................
> Device /dev/ttyACM0 has appeared; [...]

I found something that worked!

I ran the following two commands in a new terminal, but I'm pretty sure the second command alone is what did the job:

```

echo 'ATTRS{idVendor}=="2a03", ENV{ID_MM_DEVICE_IGNORE}="1"' | sudo tee /etc/udev/rules.d/77-arduino.rules

sudo udevadm trigger

```

Not entirely sure of why it worked, but I was able to flash the keyboard

```

avrdude: verifying ...
avrdude: 25244 bytes of flash verified

avrdude: safemode: Fuses OK (E:CB, H:D8, L:DE)

avrdude done.  Thank you.

```

closed 2021-02-23 by mattdibi