mattdibi/yampad #11: Multiple key presses not registering - debouncing issue?

Daveyr, 2020-10-09:
I have successfully built a yampad and all keys are working, including keys across multiple layers. However, when I rapidly press any of the keys the second and subsequent key presses are not being registered.

Is this a firmware issue or a hardware one? My instinct is that this is a debouncing setting to avoid multiple key presses when you intend only one. However, the current setting means that the numpad is not very usable.

mattdibi, 2020-10-10:
Hi,

have you tried changing the [debouncing delay](https://beta.docs.qmk.fm/using-qmk/software-features/feature_debounce_type) in QMK? Does it improve?

Daveyr, 2020-10-12:
Good call, this was the reason in the end. I used a specific debouncing method (sym_eager_pk) and changed debounce to 2 ms from 5 ms. Now the numpad is a dream to type on. All the firmware files are in https://github.com/Daveyr/yampad. I can issue a pull request if useful?

mattdibi, 2020-10-17:
Glad to hear it. I would happily accept a PR on my qmk firmware fork if you are so inclined.