mattdibi/yampad #13: LEDs not working

alytle, 2020-11-03:
Hey Mattia,

Thanks for creating this project! I had a ton of fun with my build, as it's one of my first such projects. I managed to get everything working as expected, with the exception of the RGB lights. 

I know you're busy and can't dive into fixing everyone's build, but do you have any high level suggestions on how to go about debugging the RGBs? Right now I can toggle the RGB on/off and see the status change on the OLED, but I'm getting no lights.

I had a few diodes which weren't properly secured, and those were easy to detect and repair because everything worked except for that switch, but the RGB is tougher because I'm getting nothing at all. I assume it's some sort of circuit where I have a broken connection somewhere?
[image]

These are the LEDs I used: https://www.digikey.ca/en/products/detail/sparkfun-electronics/COM-16346/11630211 and I was careful (I think) to properly align them. 

I'll include a picture of my board case you can see something obviously wrong. Feel free to mock my terrible soldering.  :)

mattdibi, 2020-11-21:
Hi,

sorry for taking so long to answer. Since the LEDs are daisy chained (see schematic for details) I think that the connection is broken at some point and preventing any of them to light up. Check with a multimeter if they're receiving power and if signal il properly reaching every one of them.

[Here](https://www.luxalight.eu/sites/default/files/styles/max_650x650/public/legacy17/276/images/WS2812B_7.jpg?itok=GeBlO494) you can find the pinout of the LEDs. Using a multimeter you should see 5V between VDD and VSS. The Dout from a LED should go to the Din of the subsequent LEDs.

Hope this helps.

alytle, 2020-11-24:
Thanks! I really appreciate the help. I'll take a stab at it on the weekend and see if I can figure out what went wrong. Hopefully it might help others.

alytle, 2020-12-02:
I had a look at the voltage across the RBGs. The VSS/VDD gives me 4.77V across all the LEDs. What's the correct way to measure DIn/Dout? I tried testing it across the individual RGBs and get inconsistent numbers, but that probably makes sense if it's supposed to be between LEDs. 

Thanks again!

mattdibi, 2020-12-04:
> What's the correct way to measure DIn/Dout? I tried testing it across the individual RGBs and get inconsistent numbers, but that probably makes sense if it's supposed to be between LEDs.

These pins are used for data so you'll need an oscilloscope to see what's actually passing between them. I suggest you to check only continuity between them and the pin that is driving them.

mattdibi, 2020-12-06:
A correction:
> Since the LEDs are daisy chained (see schematic for details) I think that the connection is broken at some point and preventing any of them to light up.
In reality the LEDs should light up to the point the chain is broken. This may suggest that the chain is broken on the very first connection (arduino pin -> first LED), I would suggest a continuity check right on this connection.

alytle, 2020-12-07:
Hey, thanks for continuing to assist. I did add some solder to F4, though it was actually better in reality than it looked. I got no lights at first, but when I used the meter to test from F4 to Din on LED0, I suddenly got lights on RGB0 & 1! So it looks like I have my voltage pins soldered correctly but I've got a break in my data pins. I added more solder to Din on RGB0 and now these two work properly, without the meter bridge.

I was able to manually turn on all the rest of the LEDs using the meter. Some came on in groups, others individually. I find it a bit strange that I was perfect with the voltage, but so bad with the data pins. Is this because the data signal flows through the caps? They are by far the hardest when you're new to soldering, so perhaps a bunch of my caps are not connected properly?

Now that I've got a heartbeat I'll keep plugging away. So far I think the following info would be useful for others who try this build and run into similar issues:
- If some keys work and others don't, that's probably the diode connected to the misoperating key, resolder it
- If the LEDs dont come on:
    - Make sure you've enabled RGB through the firmware: RGB Layer + RGB Toggle
    - Check the VSS/VDD voltage on each RLED, it should be consistent for all, regardless of whether the lights are on
    - Check the signal line, starting at F4 and going to DIn on each LED, using a bridge if needed to get them to turn on

mattdibi, 2020-12-07:
>  Is this because the data signal flows through the caps?

Nope. Look at the schematic for details.

I’m glad you found what was wrong with the board. Have fun with it ;)

closed 2020-12-07 by mattdibi