mattdibi/yampad #14: issue flashing firmware

bilsch, 2021-04-03:
Hey,

not sure how to ask in a form or anything - missed it or just not a place to ask?

Whenever I try to flash the firmware I get the following error:

<details>

```
❯ make yampad:default:avrdude
QMK Firmware 0.6.396
Making yampad with keymap default and target avrdude

avr-gcc (Homebrew AVR GCC 9.3.0_2) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  24856	      0	  24856	   6118	.build/yampad_default.hex

Copying yampad_default.hex to qmk_firmware folder                                                   [OK]
Checking file size of yampad_default.hex                                                            [OK]
 * The firmware size is fine - 24856/28672 (3816 bytes free)
Detecting USB port, reset your controller now..........
Device /dev/tty.usbmodem14101 has appeared; assuming it is the controller.
Waiting for /dev/tty.usbmodem14101 to become writable.

Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding

avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
Found programmer: Id = ""; type =
    Software Version = .; Hardware Version = .
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: leave prog mode
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: exit bootloader

avrdude done.  Thank you.

make[1]: *** [avrdude] Error 1
make: Make finished with errors
*** [yampad:default:avrdude] Error 1
```

</details>

I tried copying the file from the repo and see the same result. I'm trying with these boards:

https://www.sparkfun.com/products/15795
https://www.sparkfun.com/products/12640

Any ideas? I have the same issue from linux, mac and windows. If I use arduino to upload blink works no problem

mattdibi, 2021-04-04:
Hi,

I already encountered this behavior on Linux and usually [this](https://arduino.stackexchange.com/a/75189) solved the issue. Does this help you?

It’s weird that with the Arduino IDE works...

bilsch, 2021-04-05:
Thats interesting but I have the issue from windows, linux and mac. I was able to get the firmware to apply to the usb micro variant but the usb-c one I have not gotten working.

Essentially I triggered the reset pin and extra time and that seemed to work. Feels like a timing issue in avrdude or elsewhere to me ( and/or a bug in the pro micro from sparkfun? )

Anyway I really appreciate the response and the project thanks!

closed 2021-04-05 by bilsch