mattdibi/yampad #22: SMD or normal wire diodes

konisteel, 2024-12-08:
Hey i plan on making a pcb and is it possible to use normal wire diodes instead of smd ones? Mainly because it's hard to find it and hard to solder it.
Thanks

mattdibi, 2024-12-09:
Hello there @konisteel!

The PCB available in this repository doesn't support THT diodes (i.e. "wire diodes") only SMD ones.

closed 2024-12-09 by mattdibi

konisteel, 2024-12-09:
Can't I solder the wires in the pads? It will still work the same right?
thanks

konisteel, 2024-12-09:
band the wires and cut them so a little bit is visible and solder that in the pads
[image]

konisteel, 2024-12-09:
Or maybe i can use these?
[image]
i dont know why they look like that lol

mattdibi, 2024-12-09:
> Can't I solder the wires in the pads? It will still work the same right? thanks

Well yeah, I wouldn't recommend it but it should work.

> Or maybe i can use these?

I think that that's a more exotic form factor than the officially supported one. Are you sure you want to push your luck with those?

The SOD-123 diodes I listed in the part list should be fairly easy to solder by hand...

konisteel, 2024-12-10:
Well the SOD-123 diodes for 1n4148 is not available in my country only the wire and exotic form factor
if you have extra time maybe you could edit the file to make it fit the wire diodes by adding holes to it?

kuro-codes, 2024-12-10:
> Well the SOD-123 diodes for 1n4148 is not available in my country only the wire and exotic form factor if you have extra time maybe you could edit the file to make it fit the wire diodes by adding holes to it?

I mean it is open source and you just could add them yourself if you need them.

mattdibi, 2024-12-10:
I think that 0805 1N4148 diodes should fit inside a SOD-123 footprint and should be readily available everywhere. IIRC I used those for my PCBs. Before purchasing them give it a proper look though.

> if you have extra time maybe you could edit the file to make it fit the wire diodes by adding holes to it?

I'm sorry but I cannot do that currently.

konisteel, 2024-12-10:
> I think that 0805 1N4148 diodes should fit inside a SOD-123 footprint and should be readily available everywhere. IIRC I used those for my PCBs. Before purchasing them give it a proper look [...]

those also unfortunately are not available. Only wires and those exotic but 1 shop has them and it is double the price of normal ones so not a chance as I plan on building 20 of them