mattdibi/yampad #24: Underglow & capasitors

konisteel, 2024-12-12:
Hey i just realized you used capacitors beside the ws2812b leds. If I don't install the leds than the capacitors are not necessary right?

konisteel, 2024-12-12:
Or i can use the leds without the capacitors? will it damage the pcb or anything i need to be ware of?

konisteel, 2024-12-12:
Or can you suggest some other capacitors that i can use? like 104 100nf

[image]

mattdibi, 2024-12-13:
Hi @konisteel,

you're correct: the capacitors are only required if you install the ws2812b leds. I should have put a note in the main readme about this.

konisteel, 2024-12-13:
Will there be any issues if I use ws2812b leds without the capacitors?

mattdibi, 2024-12-14:
> Will there be any issues if I use ws2812b leds without the capacitors?

The [datasheet](https://cdn-shop.adafruit.com/datasheets/WS2812B.pdf) application circuit uses them. I would not recommend running them without capacitors but I can't say for sure whether not installing them will introduce any issue. Do it at your own risk 🤷

closed 2024-12-14 by mattdibi